pimoroni/pantilthat-python #5: Reading current servo values

pambrose, 2017-01-26:
Do you plan to implement the ability to read the current servo value?

Gadgetoid, 2017-02-06:
That's an interesting idea- any particular reason why?

It should *just work* since the values are stored as pulse widths in a pair of 16bit registers,  but they would need converted back from pulse time to degrees which could result in aliasing producing a different value to the one you sent to the driver in the first place.

Of course, the angle could also be soft-stored in code within the library, but that's usually done implicitly by the consuming program anyway.

pambrose, 2017-02-08:
It is not a must-have feature, but it would be nice to make the hat API consistent with what you get when using firmata to control a servo.

Gadgetoid, 2017-11-20:
Looks like https://github.com/pimoroni/pantilt-hat/commit/6be6df52e84c34d7d8f6385d5084eba1f40b2f5c needs to make it into a release before this issue can be considered complete.

closed 2018-02-06 by Gadgetoid