pimoroni/pantilthat-python #10: Getting angles - 'get_servo_two' returns the angle of servo_1

baychillidog, 2017-07-27:
When getting the angle of the tilt, the angle of the pan is returned.

Line 421 of [pantilt.py](https://github.com/pimoroni/pantilt-hat/blob/master/library/pantilthat/pantilt.py) of get_servo_two is:

`us = self._i2c_read_word(self.REG_SERVO1)`

to fix this should read:

`us = self._i2c_read_word(self.REG_SERVO2)`

Gadgetoid, 2017-07-28:
Drat! Good catch, thank you. Looks like I had a glaring gap in my test coverage!

robmarkcole, 2017-08-28:
Can confirm this error :-)

robmarkcole, 2017-08-31:
Any chance of a pull to fix this one? Hoping to publish https://github.com/robmarkcole/pan-tilt-phat-HASS with this functionality

robmarkcole, 2017-10-22:
ping?

baychillidog, 2017-10-23:
Thanks to @Gadgetoid  - was fixed straight away https://github.com/pimoroni/pantilt-hat/commit/6be6df52e84c34d7d8f6385d5084eba1f40b2f5c

closed 2017-10-23 by baychillidog

robmarkcole, 2017-11-01:
Any chance of releasing this as a new release on Pypi?