pimoroni/pantilthat-python #12: py3.5 install error

robmarkcole, 2017-09-04:
On attempting to install in a python 3.5 environment I receive the following error, any ideas?

```
  File "/home/pi/.homeassistant/deps/lib/python3.5/site-packages/pantilthat/pantilt.py", line 415, in get_servo_one
    return self._servo_us_to_degrees(us, us_min, us_max)
  File "/home/pi/.homeassistant/deps/lib/python3.5/site-packages/pantilthat/pantilt.py", line 131, in _servo_us_to_degrees
    self._check_range(us, us_min, us_max)
  File "/home/pi/.homeassistant/deps/lib/python3.5/site-packages/pantilthat/pantilt.py", line 120, in _check_range
    max=value_max))
ValueError: Value 51200 should be between 575 and 2325

Gadgetoid, 2017-09-04:
Could you post the code used to trigger this ValueError, ideally as a minimal example?

PanTilt HAT is passing my test coverage on Python 3.5.

Are you sure no other code is running that might be setting i2c register 1 on PanTilt HAT (the most significant byte of the 16bit millisecond pulse time) to 200? Since `200 << 8 = 51200`

Gadgetoid, 2017-11-20:
Did you find a resolution to this problem?

closed 2017-11-20 by Gadgetoid