pimoroni/pantilthat-python #29: raise IOError("Failed to read byte")

peterss7, 2021-06-29:
I have the Pimoroni pantilt hat attached to a raspberry pi 4 8gb with the neopixel stick. In the program I am working on, I use Tensorflow and Opencv to track objects. The program works, but within minutes of starting I get the error: 

OSError: failed to write byte

Earlier in the message that is spit out when the program crashes, it says "failure to read byte."  I believe it has something to do with i2c, even though I am not using i2c in my code or set up. I can provide more information, I'm just not sure what more is relevant.

druck13, 2021-06-30:
It's all done with i2c, have a look in library/pantilthat/pantilt.py

Gadgetoid, 2021-06-30:
This could be a power fluctuation or intermittent wiring issue causing i2c communications with the driver chip on PanTilt HAT to be unstable. How is your setup wired/powered?

It could equally be some quirk of the code you're driving it with since threaded code- for example- can cause issues with i2c transactions clobbering each other.

peterss7, 2021-06-30:
> This could be a power fluctuation or intermittent wiring issue causing i2c communications with the driver chip on PanTilt HAT to be unstable. How is your setup wired/powered?
>
> It could equally be [...]

......
I think it was the code.  I made a change that took a little load off of the hat and now it's working exactly like it's supposed to without crashing.

Sreehar, 2022-05-12:
> 

Hey, can you tell how you got it working in detail.
I too am facing the same problem.

chucknitro, 2023-03-23:
Same here.  Anyone help?

Sreehar, 2023-03-23:
> Same here. Anyone help?

[image]
Are you using this board exactly?
Or this 
[image]

chucknitro, 2023-03-28:
I think I found it.  The cable underneath the hat short out on the heatsink.  Added static paper and it all worked :)

Give that a try.

Julian-bt, 2023-05-22:
> > Same here. Anyone help?
> 
> [image] Are you using this board exactly? Or this [image]

Did you find the issue ?
I have the same problem...