pimoroni/pantilthat-python #31: apt-get install pimoroni does not work

jhwinters, 2022-09-22:
The README for this project suggests installing a Pimoroni dashboard with:

apt-get install pimoroni

but that package isn't in the Raspberry Pi OS repository.  Presumably it is necessary to add another repository to the apt configuration.

Gadgetoid, 2022-09-22:
It was never packaged for recent versions of Raspberry Pi OS, since I haven't had time to maintain it. Will need to update the README or get that sorted.

jhwinters, 2022-09-22:
Thank you for the prompt response.  Yes, an update to the README would seem to be called for.

If you are updating the README, you might want to include some clues about what to do after you have installed the software by the other method - curl'ing a script.  Doing that seems to get you all of the software but none of the examples will run as they stand, generating "File not found" errors when you try to run them.

The linked tutorial covers merely the physical assembly.  I'm about to start investigating the examples to see what else is missing.

Gadgetoid, 2022-09-22:
"File not found"? Could you screenshot or paste the full error? Probably requires the i2c bus to be enabled.

This neglected old repository probably needs bringing up to the state of the art for our installer scripts.

jhwinters, 2022-09-22:
Thank you - your guess was correct and saved me some investigation time.

It's these little bits missing from the instructions which can make all the difference between a smoothly working product and a frustrating amount of fiddling.

helgibbons, 2023-01-20:
This one looks to be sorted so am closing for now - feel free to reopen if not!

closed 2023-01-20 by helgibbons